Xiashangning/BigSurface #121: Fail to build due to missing `kern_util.hpp`

jc-bao, 2024-02-16:
I am trying to build the kernel without the touch screen. However I got the error: 

```cpp
#include <Headers/kern_util.hpp>
```

`BigSurface/SurfaceBattery/BatteryManager.hpp:18:10 'Headers/kern_util.hpp' file not found`

Could anybody shed a light on that?

jlempen, 2024-07-28:
@jc-bao 
I just stumbled across the same issue as you. To successfully compile `BigSurface.kext`:
>git clone --recursive https://github.com/Xiashangning/BigSurface

Then download the DEBUG version of `Lilu.kext` and `VirtualSMC.kext` and drop them into the `Dependencies` folder of the BigSurface source folder.

Now it compiles without errors on my Xcode 15.4. It took me several days to figure it out :-)

jlempen, 2024-07-28:
But if you only want to inhibit loading of the touchscreen driver, no need to build a new kext: https://github.com/Xiashangning/BigSurface/issues/102

Just delete the `SurfaceManagementEngine` item in `Info.plist -> IOPersonalities` so the touchscreen won't be loaded.

jc-bao, 2024-07-29:
Yeah, I have resolved that issue with the exact same method. Anyway, thanks for the information!

closed 2024-07-29 by jc-bao