Xiashangning/BigSurface #126: Trackpad not working on MacOS Sequoia SL3 and SB3 (and the fix for it).

ROCKERLIQUID, 2024-07-18:
Trackpad stopped working on Sequoia Developer Beta(15), probably because there were some changes under the hood.
However I found the solution by replacing the VoodooInput.kext inside the plugin folder of BigSurface with the latest version of 1.1.5 thats updated for Sequoia.

maoxixi18fowever, 2024-07-24:
> Trackpad stopped working on Sequoia Developer Beta(15), probably because there were some changes under the hood. However I found the solution by replacing the VoodooInput.kext inside the plugin folder of BigSurface with the latest version of 1.1.5 thats updated for Sequoia.
Hi, I noticed that the latest version is only 1.1.4

maoxixi18fowever, 2024-07-25:
> > Hi, I noticed that the latest version is only 1.1.4
> 
> Please check [here](https://github.com/dortania/build-repo/releases?page=1)

You're right, and it works.Thanks.

jlempen, 2024-08-07:
@ROCKERLIQUID @maoxixi18fowever 

FYI, [VoodooInput.kext v1.1.5](https://github.com/acidanthera/VoodooInput/releases/download/1.1.5/VoodooInput-1.1.5-RELEASE.zip) was officially released two days ago.