Xiashangning/BigSurface #127: Fix for Hibernation (hibernatemode 25) on IceLake Surface devices

jlempen, 2024-07-25:
Source: [Surface-IceLake-macOS-Hibernation-Fix](https://github.com/jlempen/Surface-IceLake-macOS-Hibernation-Fix)

## Surface-IceLake-macOS-Hibernation-Fix
After many hours of tinkering, **I finally found a way to fix ACPI S4 Hibernation** (`hibernatemode 25`) on the **i5 Surface Pro 7** on macOS 13.6.7 Ventura, **15-Inch i7 Surface Laptop 3** on macOS 14.5 Sonoma, **i5 Surface Laptop Go 1** on macOS 13.6.7 Ventura and very likely on the **Surface Book 3** (untested) as well. ACPI S3 Sleep (`hibernatemode 0`) is still broken, though, but perhaps the Hibernation fix will lead the way to a full S3 Sleep fix at some point in the future.

The key to the fix is to **enable ACPI S3 Sleep** in the DSDT. This is actually very easy and I'm stunned nobody tried this before. Searching for `_S3` in the `DSDT.aml` file leads us to this:

```
    If (SS3)
    {
        Name (_S3, Package (0x04)  // _S3_: S3 System State
        {
            0x05, 
            Zero, 
            Zero, 
            Zero
        })
    }
```
So to enable `_S3`, `SS3` needs to return `One`. Now searching for `SS3` leads us to this:
```
    Name (SS1, Zero)
    Name (SS2, Zero)
    Name (SS3, Zero)
    Name (SS4, One)
```

## Enable ACPI S3 Sleep
Change `Name (SS3, Zero)` to `Name (SS3, One)` to enable **ACPI S3 Sleep** with the following ACPI patch in `ACPI -> Patch`:
```
Comment: Rename Name (SS3, Zero) to One - Enable S3 Sleep
Find: 5353335F00
Replace: 5353335F01
```
Before the rename, the system log shows:
> (AppleACPIPlatform) ACPI: sleep states S4 S5

After the rename, the system log shows:
> (AppleACPIPlatform) ACPI: sleep states S3 S4 S5

and commands such as `pmset -g log` or `pmset -g log | grep -e "Sleep.*due to" -e "Wake.*due to"` now clearly show that the device is entering sleep and (eventually) waking up from sleep.

## Add Reserved Memory region
This fix was found in [Tyler Nguyen's x1c6-hackintosh repo](https://github.com/tylernguyen/x1c6-hackintosh/issues/44#issuecomment-705971765). Thank you @benbender @1Revenger1 @savvamitrofanov and @vit9696!

Get rid of the black screen on Wake with the following Reserved Memory region in `NVRAM -> ReservedMemory`:
```
Comment: Fix hibernate mode 25 black screen on wake
Address: 569344
Size: 4096
Type: RuntimeCode
Enabled: true
```

## Enable DiscardHibernateMap
This doesn't seem to be required on every Surface device, but at least on the Surface Laptop 3, it fixes a kernel panic on Wake.

Enable the following Booter Quirk in `Booter -> Quirks`:
```
DiscardHibernateMap=true
```

## Set HibernateMode to NVRAM
In `Misc -> Boot`, set `HibernateMode` to `NVRAM`:
```
HibernateMode=NVRAM
```

## Add the GPRW instant wake patch
At least on the **Surface Pro 7** and the **Surface Laptop Go 1**, the GPRW instant wake patch is required as well. 

Add [SSDT-GPRW.aml](https://github.com/jlempen/Surface-IceLake-Hibernation-Fix/blob/main/SSDT-GPRW.aml) to the ACPI folder of your EFI. Then add `SSDT-GPRW.aml` to `ACPI -> Add` and the following ACPI patch to `ACPI -> Patch` in the `config.plist` file:
```
Comment: Change Method(GPRW,2,N) to XPRW, pair with SSDT-GPRW.aml
Find: 4750525702
Replace: 5850525702
```

## Enable hibernatemode 25 in macOS
Open the `Terminal` and enter the following commands, then reboot for the changes to take effect:
```
sudo pmset restoredefaults
sudo pmset -a hibernatemode 25
```
If for whatever reason Hibernate is still not working on your system, you should reset the `Power Management` settings and rebuild the `sleepimage` file. To do so, open the `Terminal` and enter the following commands, then reboot for the changes to take effect:
```
sudo rm /Library/Preferences/com.apple.PowerManagement*
sudo rm /var/vm/sleepimage
sudo pmset hibernatefile /var/vm/sleepimage
```
Once you are back in macOS, disable Sleep and enable Hibernate again, then reboot:
```
sudo pmset restoredefaults
sudo pmset -a hibernatemode 25
```

## Fix broken Bluetooth on Wake from Hibernation
After the device wakes up from Hibernation, Bluetooth may be broken / unable to connect.

A very simple fix for this issue is to [download and install Bluesnooze](https://github.com/odlp/bluesnooze). Launch the app, enable `Launch at login` and you're done!

## A few comments / Help needed!
Hibernation seems to be working perfectly on my i5 / 8GB / 256GB **Surface Pro 7** and i5 / 8GB / 256GB **Surface Laptop Go 1** both running on macOS Ventura 13.6.7 with the SMBIOS `MacBookAir9,1` and using [Xiashangning's BigSurface.kext v6.5](https://github.com/Xiashangning/BigSurface/releases/tag/v6.5). Everything comes back online after waking up from hibernation.

However, my i7 / 16GB / 2TB WD SN770M **15-Inch Surface Laptop 3** running macOS Sonoma 14.5 with the SMBIOS `MacBookPro16,2` and with [Xiashangning's BigSurface.kext v6.2](https://github.com/Xiashangning/BigSurface/releases/tag/v6.2) wakes up with a dead trackpad. I have yet to find a way to fix the dead trackpad.

The main difference between the **Surface Pro 7** or **Surface Laptop Go 1** and the **Surface Laptop 3** or **Surface Book 3** is that on the former, the keyboard and trackpad are attached through USB, whereas on the latter, they are attached through a proprietary interface.

Both the **Surface Laptop 3** and the **Surface Book 3** are plagued by a nasty issue with the trackpad on macOS. Fixing the issue is possible, but [requires downgrading the firmware to 13.101.140.0 and BigSurface.kext to 6.2](https://github.com/Xiashangning/BigSurface/issues/79#issuecomment-2208484390), that's why it still runs with `BigSurface.kext` v6.2.
The trackpad works with the latest `BigSurface.kext` v6.5, though, but the trackpad lags and skips every few seconds. Furthermore, tests show that the `BigSurface.kext` v6.5 doesn't fix the dead trackpad after wake from hibernation.

Playing around trying to unload the `BigSurface.kext` or `VoodooSerial.kext` plug-in before hibernate and reloading it on wake from hibernate has been a dead end until now, as I'm unable to do so due to various errors with the `kextstat`, `kextunload` and `kextload` commands in macOS.

So, the first steps to troubleshoot the trackpad issue would be to downgrade macOS to 13.6.7 Ventura and change the SMBIOS back to `MacBookAir9,1`, but I must admit I'm not all too confident that would fix the dead trackpad after wake from hibernation. We would probably need to find someone who is able to fix the `BigSurface.kext` and I'm not sure @Xiashangning is still maintaining his `BigSurface` repo, as there hasn't been any activity for over a year now, [since July 14th 2023](https://github.com/Xiashangning/BigSurface/issues/109#issuecomment-1636433545).

jlempen, 2024-07-31:
So I fixed the dead trackpad on wake from hibernate on the `Surface Laptop 3` and very likely on the `Surface Book 3` as well.

The issue was actually an issue in VoodooI2CHID which was fixed by https://github.com/VoodooI2C/VoodooI2CHID/commit/89fa5b64c1fc38b4b370d45c7ad0d8d632e3b86c

Now the interesting thing is that the latest `BigSurface v6.5` forked and built with the above commit fixes the dreaded [Trackpad lagging/skipping bug](https://github.com/Xiashangning/BigSurface/issues/79) as well after the first hibernate/wake cycle following the laptop's boot into macOS.

What this means is that in theory, `Surface Laptop 3` and `Surface Book 3` users should now be able to update their firmware through Windows Update and still have a smooth trackpad experience on macOS, provided they go through a hibernate/wake cycle after booting macOS. This still requires thorough testing, though, and I'm quite confident I should be able to fix this issue in the coming weeks.

I also built the older `BigSurface v6.2` with the above commit and with this version, the `Surface Laptop 3` has a buttery smooth trackpad right from startup and also after wake from hibernate, provided the user follows [this procedure](https://github.com/Xiashangning/BigSurface/issues/79#issuecomment-2208484390) to downgrade the firmware of their Surface Laptop 3 or Surface Book 3.

I would be glad if a few `Surface Laptop 3` and `Surface Book 3` users could try those test builds and report back.

[BigSurface62SL3.zip](https://github.com/user-attachments/files/16448021/BigSurface62SL3.zip)
[BigSurface65SL3.zip](https://github.com/user-attachments/files/16448027/BigSurface65SL3.zip)

jc-bao, 2024-08-01:
@jlempen Amazing work! Will let you know after testing it out on SL3.

ROCKERLIQUID, 2024-08-01:
Doing god's work for our devices, thanks a lot man.

jlempen, 2024-08-06:
@ROCKERLIQUID 

> ReservedMemory is in UEFI, not NVRAM as per your guide.
>

Corrected, thanks!

> I tried your kext v6.5, I had to update the VooodoInput.kext with the new patch for Sequoia, and the trackpad and keyboard does not work after a hibernate cycle. 
> 

Thanks for the information. I'll have a look at it later on, once Sequoia goes out of Beta and once I've fixed the trackpad issues on SL3 and SB3 running Ventura and Sonoma.

But it's already good to know that hibernation seems to work on your device running Sequoia Beta!

BTW, is your Surface a Laptop 3 or a Book 3?

ROCKERLIQUID, 2024-08-06:
@jlempen 

> BTW, is your Surface a Laptop 3 or a Book 3?

Surface Laptop 13" i5-1035G7.

jc-bao, 2024-08-06:
@jlempen Just tried your method on my SL3 i7 machine with firmware 6.2 and Ventura. It works perfectly! One thing I'm not so sure about is the instant wake-up. For me, I still have to wait a few seconds before it wakes up. And when waking from sleep, I see the Microsoft logo. I'm not sure if that's normal, although I have included GPRW. I'm curious about your experience with SL3. [Here](https://github.com/jc-bao/surface-laptop3-ventura) is my EFI.

Another minor issue I'm experiencing is that the machine sometimes overheats, which never happened on my i5 SL3. I'm wondering if @jlempen @ROCKERLIQUID  can control the fan speed or change the power plan.

Overall, kudos to @jlempen for the great work!

jlempen, 2024-08-06:
@jc-bao 

> One thing I'm not so sure about is the instant wake-up. For me, I still have to wait a few seconds before it wakes up. And when waking from sleep, I see the Microsoft logo. I'm not sure if that's normal
> 

Hibernate is not like sleep: when your device goes into hibernate mode, it saves the content of the RAM to a file on the SSD and turns off completely. That's why you have to press the power button to wake it up. Then it goes through the normal startup sequence (MS logo with red padlock), then OpenCore silently launches macOS which reloads the content of the RAM from the hibernation file on the SSD. This all takes around 10 seconds.

I'm afraid we will probably never going to fix normal sleep with instant wakeup for Surface devices, as Micro$oft does not support ACPI S3 sleep anymore.

So you can either enable `hibernatemode 25` with 0% battery drain during hibernation but 10 seconds to wake up, or enable `hibernatemode 0` with a battery drain of around 10% per hour but instant wake working, as it basically only turns off the display... 

> Another minor issue I'm experiencing is that the machine sometimes overheats, which never happened on my i5 SL3. I'm wondering if @jlempen @ROCKERLIQUID  can control the fan speed or change the power plan.

I have both an i5 13" SL3 and an i7 15" SL3 and have no overheating issues on neither device, but it would be worth investigating if changing the SMBIOS from `MacBookPro16,2` to `MacBookAir9,1` prevents your i7 from overheating. I also run my devices with the `Low Power Mode` preference in macOS set to `Always`.

You could also make your own `CPUFriendDataProvider.kext` and optimize power saving settings with the help of CorpNewt's CPUFriendFriend app.

BTW, when the Surface devices are charging, they heat up quite a lot and thus the fan is almost always spinning.

I don't know if there's a way to control the fan, I'll have to investigate this as well 😀

jlempen, 2024-08-06:
@jc-bao 
I just checked my EFI for the i7 SL3, I also use the SMBIOS `MacBookPro16,2` like you, so your overheating issue is apparently not related to the SMBIOS...

jc-bao, 2024-08-06:
@jlempen, thank you so much for the information! I think hibernate is enough to make the machine usable for daily use. I think `Low Power Mode` is definitely a solution. I don't know why my machine sometimes goes up to 40W when plugged in, but overall, the Hackintosh is almost perfect thanks to your efforts. I really appreciate the great job you've done!

ROCKERLIQUID, 2024-08-06:
> Another minor issue I'm experiencing is that the machine sometimes overheats, which never happened on my i5 SL3. I'm wondering if @jlempen @ROCKERLIQUID can control the fan speed or change the power plan.

I live in South Asia where the climate is almost always hot, and my Surface always throttles in Windows and in MacOS, the dreaded 0.4ghz lock when BD-PROCHOT gets triggered even under light load.

Solution was to download a BD-PROCHOT disabling EFI driver and it works beautifully ever after. Ofcourse, battery life gets a huge hit which was never good on MacOS anyways.

And don't worry the laptop still shuts down if it gets too hot, but I have never experienced it as MacOS knows how to handle these CPUs well.

jc-bao, 2024-08-06:
> Solution was to download a BD-PROCHOT disabling EFI driver and it works beautifully ever after. Ofcourse, battery life gets a huge hit which was never good on MacOS anyways.

I see. 😂 Seems it is a common issue for surface. Thanks a lot for the solution!

tashda, 2024-08-07:
I have tried this with my SL3 i7, and have tried to follow all steps, but when returning from hibernation the screen "glitches", pixelated and in inverted colors, and then turn off.

I have tried a lot of configurations to be able to output image through the USB-C port (to a dock connected with DisplayPort) so I am looking in that direction as I have a lot of various WhateverGreen entries.

As @jc-bao have shared EFI, I am currently matching my own configuration with this. But one question before I do a deep dive into my own EFI - does display output work on your machines?

I have downgraded the SL3 firmware and am currently running the BigSurface62 kext.
What is currently working (which I have struggled with) is Display output through USB-C
What is not working is waking from hibernation (and also not able to boot with USB-C display output connected)

jlempen, 2024-08-07:
> I tried your kext v6.5, I had to update the VooodoInput.kext with the new patch for Sequoia, and the trackpad and keyboard does not work after a hibernate cycle.

I updated OC and all kexts a few moments ago and run into the same issue now. So it's not related to Sequoia, but rather to some other kext, perhaps Lilu or WhateverGreen, I'll have to investigate...

jlempen, 2024-08-07:
@tashda 

> I have tried this with my SL3 i7, and have tried to follow all steps, but when returning from hibernation the screen "glitches", pixelated and in inverted colors, and then turn off.
> 

Is this happening on the internal display of your SL3? The pixelated and inverted colors can happen sometimes on wake from hibernate, but usually it's simply black until the lock screen appears. Is the device really turning off, or is the screen just black? You can easily test this by enabling "Play feedback when volume is changed" and hitting the volume keys when the display turned off. Does the `Fn` key on the keyboard light up when you press it once your display turns off?

Perhaps there's something wrong with your ReservedMemory patch under `UEFI -> ReservedMemory`:
```
Comment: Fix hibernate mode 25 black screen on wake
Address: 569344
Size: 4096
Type: RuntimeCode
Enabled: true
```

If that's not it, I would suggest clearing and cleaning your NVRAM and restoring the sleepimage file:

## Enable hibernatemode 25 in macOS
Open the `Terminal` and enter the following commands, then reboot for the changes to take effect:
```
sudo pmset restoredefaults
sudo pmset -a hibernatemode 25
```
If for whatever reason Hibernate is still not working on your system, you should reset the `Power Management` settings and rebuild the `sleepimage` file. To do so, open the `Terminal` and enter the following commands, then reboot for the changes to take effect:
```
sudo rm /Library/Preferences/com.apple.PowerManagement*
sudo rm /var/vm/sleepimage
sudo pmset hibernatefile /var/vm/sleepimage
```
Once you are back in macOS, disable Sleep and enable Hibernate again, then reboot:
```
sudo pmset restoredefaults
sudo pmset -a hibernatemode 25
```

> I have downgraded the SL3 firmware and am currently running the BigSurface62 kext. What is currently working (which I have struggled with) is Display output through USB-C What is not working is waking from hibernation (and also not able to boot with USB-C display output connected)

Once you get hibernation working, please update to my test version of BigSurface v6.2 which fixes the dead trackpad on wake from hibernate:
https://github.com/Xiashangning/BigSurface/issues/127#issuecomment-2261648714

jc-bao, 2024-08-07:
> As @jc-bao have shared EFI, I am currently matching my own configuration with this. But one question before I do a deep dive into my own EFI - does display output work on your machines?

@tashda My built-in display works well after wake up. But there are two remaining issues: 1. when i connect to external display with type-c, i have to close the lid and reopen it to make built-in screen work, otherwise i can only see the external display. 2. i also have a surface dock but the mini-dp doesn't work. I am curious if you have done any specific configuration for that? I think people have reported that mini-dp on surface dock does not work, wondering what's your work around. 

btw, my config is based on 6.2 with old firmware so maybe that could also be one factor?

jc-bao, 2024-08-07:
> I have tried a lot of configurations to be able to output image through the USB-C port (to a dock connected with DisplayPort) so I am looking in that direction as I have a lot of various WhateverGreen entries.

I can just confirm this just happened to me. I find close the lid and reopen it could work. But it does not happen every time. So weird. Never have this issue with old whatevergreen and ventura.

tashda, 2024-08-07:
I am running Sonoma 14.6. I am looking at your EFI for inspiration - copying EFI from others is the worst gift you can get yourself :)

I am currently running @jlempen 6.2 BigSurface kext, with downgraded firmware, and have the correct ReservedMemory patch.

The biggest problems I am experiencing right now
1) Kernel panic upon boot 1st time it boots, second time is OK
2) When I close the lid and open back up and press power, the screen is black - the picture then quickly shows, and is black again.
When I am not doing anything on the computer and it hibernates, it works fine again when I press the power button (with the inverted pixelated screen at first).

As I am no expert in these types of things, I am currently trying to at least make the pc work again after lid has been closed and opened again - I suspect WhateverGreen to be the cause of this, but when I tried to remove dpcd-max-link-rate from my config.plist (data 14000000), it panicked when booting, so I am still looking.

jc-bao, 2024-08-07:
@tashda 
1. For the Kernel panic thing in the first boot, I can confirm that. Wondering if @ROCKERLIQUID has encountered the same issue. 
2. Actually what I did is press the power button, wait for a while until it boots with blank screen, then close the lid for about 2s, then open it, it shall work. That is so weird. 

To me, the best choice would be running Ventura which is way more stable.

ROCKERLIQUID, 2024-08-07:
> For the Kernel panic thing in the first boot, I can confirm that. Wondering if @ROCKERLIQUID has encountered the same issue.
Actually what I did is press the power button, wait for a while until it boots with blank screen, then close the lid for about 2s, then open it, it shall work. That is so weird.

tashda, 2024-08-07:
> Actually what I did is press the power button, wait for a while until it boots with blank screen, then close the lid for about 2s, then open it, it shall work. That is so weird.

Gotcha! When I do this, it works. Can connect display through USB-C immediately after hibernation, but when I do the "double dip" of the lid it works on internal display as well.

Working here as well now :) 
Good to know there's at least 4 of us still using SL3 with OC

jlempen, 2024-08-07:
@tashda @jc-bao @ROCKERLIQUID 
I think it's time we start with a common base, so I'm setting up a triple boot Ventura-Sonoma-Sequoia Beta 5 on my i5 Surface Laptop 3 13" with the stock 256GB SK Hynix SSD running on the latest firmware from Microsoft and my test version of BigSurface v6.5 with the VoodooInput.kext submodule updated to v1.1.5 so it works on Sequoia as well. 

Then, probably tomorrow, I'll send you guys my EFI which doesn't give me any trouble with kernel panics on start (yet).

Right now, Ventura is working nicely but with a choppy trackpad, working hibernation and broken keyboard/trackpad on wake from hibernate. I'm almost sure that the broken things are due to the fact that I didn't downgrade the UEFI firmware on this machine yet. I'm really starting to think that the only reliable way to have the SL3 and SB3 working perfectly is by downgrading the UEFI to the last known good version as explained [in my thread here](https://github.com/Xiashangning/BigSurface/issues/79#issuecomment-2208484390).

I remember that I had a few kernel panics on reboot every time I would change something to my EFI folder, but usually the panic would go away after a second reboot.
I use a single EFI for both my i5 13" and i7 15" Surface Laptop 3's and I don't know if this EFI supports an external display through the USB-C port, as I don't have the hardware to test.

jlempen, 2024-08-11:
@ROCKERLIQUID 

> I tried your kext v6.5, I had to update the VooodoInput.kext with the new patch for Sequoia, and the trackpad and keyboard does not work after a hibernate cycle.

Please try one of my updated BigSurface v6.2 or v6.5 test builds from this comment:
https://github.com/Xiashangning/BigSurface/issues/127#issuecomment-2261648714

The updated versions include the VoodooInput.kext v1.1.5 for macOS Sequoia Beta 5 compatibility

If your trackpad and keyboard don't work after a hibernate cycle, it's either because you tried one of my older test build which included the older VoodooInput.kext v1.1.3, your SL3's firmware hasn't been dowgraded to UEFI firmware version `13.101.140.0` (or Windows auto-updated the firmware again).

jlempen, 2024-08-11:
@ROCKERLIQUID @jc-bao @tashda 

So this week-end I set up a Surface Laptop 3 and a Surface Book 3 each with a Ventura-Sonoma-Sequoia triple-boot to test my updated test versions of BigSurface v6.2 and v6.5:

[BigSurface v6.2 for SL3 and SB3.zip](https://github.com/user-attachments/files/16576290/BigSurface.v6.2.for.SL3.and.SB3.zip)

[BigSurface v6.5 for SL3 and SB3.zip](https://github.com/user-attachments/files/16576291/BigSurface.v6.5.for.SL3.and.SB3.zip)

Here's the conclusions of my tests:

- Both the `Surface Laptop 3` and the `Surface Book 3` need to have their UEFI firmware downgraded to version `13.101.140.0` in order to have a working trackpad and keyboard upon wake from hibernation
- Hibernation works flawlessly on both devices with Ventura, Sonoma and Sequoia Beta 5
- On both devices, the trackpad is smooth before and after hibernation with BigSurface v6.2
- On both devices, the trackpad is choppy before and smooth after hibernation with BigSurface v6.5

In a nutshell, downgrade the UEFI firmware to `13.101.140.0` and use my fork of `BigSurface` v6.2 on macOS Ventura, Sonoma or even Sequoia.

The latest Beta 5 of macOS Sequoia works quite well, but you'll have to use `itlwm.kext` v2.3 and the `HeliPort` app to get Wifi working.

I'm still trying to fix the choppy trackpad on boot with `BigSurface` v6.5.

For some odd reason, the only way to install macOS on the `Surface Book 3` running on the latest UEFI firmware is by disabling graphics acceleration by adding the `-igfxvesa` boot argument to my config.plist or the display would crap out when booting the installer. Once installed, I could enable graphics acceleration again by removing the `-igfxvesa` boot-arg.

jc-bao, 2024-08-11:
@jlempen Great job! I am wondering do you also have the same kernel panic when boot for the first time to the Sonoma as @tashda and I have experienced?

Xiashangning, 2024-08-16:
Hi guys, first I would like to thank all of you for your continuous passion in hackintosh for Surface! @jc-bao @ROCKERLIQUID @tashda, and especially @jlempen. It's true that I am unfortunately not in procession of Surface Pro 7 for almost a year :( However, as now I managed to once again get my hands on my old surface device back home, I will first take a look at all issues and your conversations above to get myself updated. Then I will see what I can do.

Xiashangning, 2024-08-16:
Also, would you mind if I invite you as the collaborator of this repo? @jlempen So that you can directly contribute to this project for Surface Laptop devices!

jlempen, 2024-08-16:
Hey @Xiashangning, very glad you're back! I hope you are well!

> Hi guys, first I would like to thank all of you for your continuous passion in hackintosh for Surface! 

Thank YOU for your tremendous work on the `BigSurface` kext!

> Also, would you mind if I invite you as the collaborator of this repo? @jlempen So that you can directly contribute to this project for Surface Laptop devices!

Please do, I'd be very happy to contribute to your repo. I'm by no means a programmer like you, and `BigSurface` is a very complex piece of code. But by playing around and bisecting the code I found a few ways to improve the macOS experience on some Surface devices.
In a nutshell, here's what's been improved since the release of `BigSurface` v6.5:

- Fix for the lagging/skipping trackpad on `Surface Laptop 3` and `Surface Book 3` (https://github.com/Xiashangning/BigSurface/issues/79#issuecomment-2208484390)
- Hibernation is now working on all Surface devices (https://github.com/Xiashangning/BigSurface/issues/127)
- Fix for all Surface devices to make the trackpad work on the latest macOS Sequoia (https://github.com/Xiashangning/BigSurface/issues/126)

As macOS Sequoia could very well be the last version of macOS running on Intel machines, my aim is to provide a version of `BigSurface` which works on all or at least all the latest (i.e. IceLake) Surface devices running macOS Sequoia.

Right now, `Surface Pro 7` and `Surface Laptop Go` devices work perfectly on Ventura, Sonoma and Sequoia with `BigSurface` v6.5, as long as the `VoodooInput.kext` submodule is replaced with [the latest v1.1.5 from Acidanthera](https://github.com/acidanthera/VoodooInput/releases/tag/1.1.5). 

`Surface Laptop 3` and `Surface Book 3` devices work perfectly on Ventura, Sonoma and Sequoia, provided that their UEFI firmware is downgraded to the last known working version `13.101.140.0` following [this procedure](https://github.com/Xiashangning/BigSurface/issues/79#issuecomment-2208484390) and the `BigSurface` kext is downgraded to v6.2. 

Moreover, I just discovered that the skipping/lagging trackpad issue on the `Surface Laptop 3` and `Surface Book 3` is due to the upgraded `SurfaceSerialHub` and `SurfaceSerialHubDevices` code in commit https://github.com/Xiashangning/BigSurface/commit/538bf71cdb1984c28c7975b0549f2a23e7f63349. 
Building `BigSurface` v6.5 with [this PR](https://github.com/Xiashangning/BigSurface/pull/128) merged and the `SurfaceSerialHub` and `SurfaceSerialHubDevices` code downgraded to the code from commit https://github.com/Xiashangning/BigSurface/commit/8fe7f68ca885c337ea5c495ebca180f1bb2c1d03 works perfectly on my SL3 and SB3 and allows users to enjoy your touchscreen improvements by installing your [IPTSDaemon v1.4](https://github.com/Xiashangning/IPTSDaemon/releases/tag/v1.4).

My ultimate goal would be to find which piece of code in `SurfaceSerialHub` and `SurfaceSerialHubDevices` is responsible for the skipping/lagging trackpad issue on SL3 and SB3 and why those Surface devices have a dead trackpad and keyboard after wake from hibernation if their UEFI firmware is more recent than `13.101.140.0`. Finding answers to these questions would make it easier for non hackintosh-savvy users to enjoy macOS on their SL3 or SB3, as they wouldn't need to downgrade their UEFI firmware and use a special build of `BigSurface`.

Anyway, welcome back @Xiashangning and thanks for all your help and great work!

jlempen, 2024-08-16:
> @jlempen Great job! I am wondering do you also have the same kernel panic when boot for the first time to the Sonoma as @tashda and I have experienced?

@jc-bao 
I have not experienced any kernel panic on boot on my SL3 and SB3. I did experience a few random kernel panics when browsing for a while in Firefox. Those KP's were fixed by using the `igfxfw=2` Whatevergreen boot-arg to force loading the Apple GuC firmware.

Did you have a look at the panic log which could give you some hints on what's causing the panic? When you reboot after the kernel panic, macOS will ask you if you want to create and send a panic log to Apple. By clicking on `Report...` you may view the panic log and then you can dismiss sending the report to Apple.

But I would suspect either an unsupported SSD (perhaps you have a Samsung SSD) or something wrong with your iGFX settings.

Please have a look at my config.plist for the 15-Inch i7 Surface Laptop 3:
[EFI_SL3_i7_15inch.zip](https://github.com/user-attachments/files/16641774/EFI_SL3_i7_15inch.zip)

leoyuhle, 2024-08-27:
I have a Surface laptop go, sleep works well with AOAC patch, also by adding the boot-arg igfxfw=2; but another issue is the display output, when plugging the cable, the pc just reboot

rongchuan17, 2025-01-20:
Hello everyone,can we find a way to fix (hibernatemode 25) on the  Surface Pro 5,？

MirComputers, 2025-03-16:
Does undervolt with voltageshift works? I have a surface go 2 and it is a lifesaver

jlempen, 2025-03-17:
@MirComputers As Overclocking Lock is enabled in the UEFI BIOS of those devices, voltageshift does not work. These UEFI BIOSes are encrypted and I haven't found a way to modify UEFI variables on them.
The Surface Go 2 is a very different machine and it has a more standard UEFI BIOS.