Xiashangning/BigSurface #132: Surface book 2

ErikLavr, 2024-10-21:
Hello! I wanted to ask if the Surface book 2 is supported. I've been trying to find anything to install it on the Surface book 2 and sadly never got it to work. And since the project page says the minimum is a Surface Book 3 but one issue on here said something about the 2nd one I just wanted to ask. Thanks!

-Erik :)

Xiashangning, 2024-10-21:
It should be supported. I have worked with some SB2 users back when we were supporting older Surface devices.
First you should make sure that your device can properly boot into macOS without BigSurface. Then, we can try to figure out why BigSurface isn't working.

ErikLavr, 2024-10-21:
Like I said I never tried bigsurface since I just found it, but thanks.

Xiashangning, 2024-10-21:
So your device runs macOS properly without BigSurface right now, is that correct?
If so, follow the guide in the README :) SB2 is supported (battery, touch screen, keyboard&touchpad), though there exists some unfixable glitches.