Xiashangning/BigSurface #133: Inquiry About New Release Plans After v6.5

mohsenV1980, 2024-11-30:
Hi,

Thank you for maintaining this project! I noticed that the last release, version 6.5, was published around February 2023. I wanted to check if there are any plans for a new release soon?

Looking forward to any updates on this.

Best regards,
Mohsen

Xiashangning, 2024-12-18:
Hi, it might comes out later this month or at the beginning of 2025:)
There has been quite a lot updates on the upstream projects and I would like to rewrite the touch logic to support better finger&palm recognition, so that took a bit longer than I thought.

yassgityass, 2025-01-25:
> Hi, it might comes out later this month or at the beginning of 2025:) There has been quite a lot updates on the upstream projects and I would like to rewrite the touch logic to support better finger&palm recognition, so that took a bit longer than I thought.

Hi 
i hope there will be an update for touchscreen on Surfacebook1
as with version 6.5 its not supported

manumr816, 2025-01-29:
when are releasing latest version bro ,
please include palm rejection in bro

manumr816, 2025-01-29:
> Hi, it might comes out later this month or at the beginning of 2025:) There has been quite a lot updates on the upstream projects and I would like to rewrite the touch logic to support better finger&palm recognition, so that took a bit longer than I thought.

when you releasing updates bro