Xiashangning/BigSurface #141: [SL3] Touchscreen driver creates stability and performance issues

ConwayTech-Dev, 2025-08-21:
As discovered by @ikador in [this issue](https://github.com/jlempen/Surface-Laptop-3-OpenCore/issues/23), the touchscreen driver completely breaks the Surface Laptop 3's ability to perform under load. The touchscreen driver results in extreme overheating and forces the system to throttle to a point where it is unusable. For some reason, it breaks the CPU/fan/thermal system completely. With the touchscreen driver enabled, the SL3 would stall and shut down due to high temperatures when doing things like lightweight gaming or performance benchmarks.

I can't say about other Surface models, but as a Surface Laptop 3 user, this issue made the laptop unusable under any somewhat-intense circumstances. This issue can be fixed by disabling the touchscreen driver.

Thanks for creating BigSurface, it is amazing and has been very influential for the Hackintosh community! And we should all hope that this issue can be resolved.

KoalafiedDev, 2025-12-17:
On my Surface Laptop 3, I ran into severe thermal throttling and UI lag under macOS: fans ramping to max, high kernel_task CPU usage, and stuttery system animations. Once temperatures dropped, everything immediately returned to normal.

After a lot of testing, I found that this behavior only occurs when SSDT-PLUG is enabled. Removing SSDT-PLUG (including a freshly generated one based on the Dortania guide) completely resolves the issue: temperatures stay stable, fans behave normally, and performance remains smooth. Re-enabling SSDT-PLUG reliably brings the problem back.

I initially suspected other components, but the issue consistently tracks back to SSDT-PLUG specifically.

Could SSDT-PLUG be the root cause in your case as well, rather than the touchscreen driver?

ConwayTech-Dev, 2025-12-17:
Hi @KoalafiedDev,

Thanks so much for bringing this up! Seems like we have the same issue. The thermal issues you're having sound like the PROCHOT symptoms that I'm also getting.

I've never really thought about the SSDTs as being causes of the thermal issues, but you might be onto something big. I'll do some testing with SSDT-PLUG and see if it's the real culprit. It definitely might be driving force behind my issues — even after removing the touchscreen driver, I was running into the severe thermal throttling (much less, though).

I'll bring this up in @jlempen's SL3 Hackintosh repository and let you know what happens with my testing. I think we mostly marked this issue as resolved after discovering the touchscreen issue thing, but further improvements would be great. Thanks again!

closed 2026-10-24 by colinvkim